phdussud/pico-dirtyJtag #1: Build error

zoobab, 2021-06-05:
I am trying to build your firmware with pico-sdk, and I hit the following error:

```
[ 26%] Building C object CMakeFiles/dirtyJtag.dir/get_serial.c.obj
In file included from /code/pico-dirtyJtag/get_serial.c:27:
/code/pico-sdk/src/rp2_common/pico_unique_id/include/pico/unique_id.h:72:52: error: unknown type name 'uint'; did you mean 'int'?
   72 | void pico_get_unique_board_id_string(char *id_out, uint len);
      |                                                    ^~~~
      |                                                    int
make[2]: *** [CMakeFiles/dirtyJtag.dir/build.make:102: CMakeFiles/dirtyJtag.dir/get_serial.c.obj] Error 1
make[2]: Leaving directory '/code/pico-dirtyJtag/build'
```

I am using pico-sdk version 1.2.0 (bfcbefafc5d2a210551a4d9d80b4303d4ae0adf7).

phdussud, 2021-06-05:
Interesting. This error does not appear with an older (1.1) version of the SDK. We need a work around...
Thanks,
Patrick.

phdussud, 2021-06-05:
I think gcc does not define uint anymore. 
The fix would be to define it right before the failing include in get_serial.c like so(change in bold)
**typedef unsigned int uint;**
#include "pico/unique_id.h"

phdussud, 2021-06-05:
I updated with the version 1.2 of the sdk and it builds with the change. I haven't tested the resulting binary yet.

phdussud, 2021-06-05:
A better fix is to replace 
#include <stdint.h>  by #include "pico/types.h". This header file defined uint the same way I indicated. 

This will give us:
#include "pico/types.h"
#include "pico/unique_id.h"

phdussud, 2021-06-05:
I committed a fix.

closed 2021-06-05 by phdussud