phdussud/pico-dirtyJtag #9: USB Port Access

DevanshuGajjar, 2023-01-18:
Hi, 
I appreciate the effort that you gave while developing the code and everything is well explainable, but actually i am new with jtag as well as rpi pico, so after loading the uf2 file in the pico i lost the usb access to my laptop, so can you please help me with that .Also will i be needing to connect any jtag device on the other hand, currently i am using the analyzer to observe the jtag frames but even after writting to the buffer i am not able to see the frames, so also if some help with this is possible it will mean a lot.

Thanks
Devanshu

jeanthom, 2023-01-18:
Hi,

What do you mean by "losing USB access" to your laptop? Do you mean that the USB controller got "borked" right after flashing DirtyJTAG?

If so, could you provide us with your hardware configuration (laptop model, operating system, etc.) and if possible any `dmesg` output happening after programming your RPi Pico?

Thanks,
Jean.

DevanshuGajjar, 2023-01-18:
Hi,

Yes I think that usb controller is getting break basically, I want to make the message to get printed using the usb and also to communicate continuously with the rpi pico also after flashing it with the dirtyJTAG. 

Hardware Details:
laptop model : dell latitude 5420
operating system : Linux (on windows)

Also can please give me some advice on the second part regarding the "Also will i be needing to connect any jtag device on the other hand? , currently i am using the analyzer to observe the jtag frames but even after writting to the buffer i am not able to see the frames, so also if some help with this is possible it will mean a lot."

**DMESG**
[image]

Thanks

phdussud, 2023-01-18:
@DevanshuGajjar: sorry about your problems. Are you using wsl2? As far as I know only COM and mounted mass storage will be forwarded to wsl from the host(Windows). DirtyJtag USB interfaces will not be visible or accessible by wsl2. 
My advice would be to plug the rp board and look at the windows device manager to see if you see  a DirtyJtag in the "other devices" category. You know you must use Zadig to change the driver for DirtyJtag to WinUSB, right? 
The other thing I want to ask is this: If you press the boot button while you connect your rp board, do you see the mass storage in Windows (and also in wsl?) This part is handled in the rp internal ROM so it still should work.

About observing some JTAG frames: You don't need a device. For example if you use OpenFPGALoader --detect -c dirtyJtag you will see quite a few JTAG frames (CLK, TMS, TDI)

DevanshuGajjar, 2023-01-30:
@phdussud the problem was resolved, Thanks for your support. Actually I wanted to make Jtag Debugger so is there anyway to use your repo for the same? I am trying with bit banging, I am able to create the fsm but the data and the clocks are not synchronizing. Can you please guide me on same ?

phdussud, 2023-01-30:
If you want to use it as a debugger, check out my clone of [openocd ](https://sourceforge.net/u/phdussud/openocd/ci/master/tree/)

phdussud, 2023-02-12:
I am closing this issue as it seems to be resolved

closed 2023-02-12 by phdussud