phdussud/pico-dirtyJtag #17: Compiled from sources... openFPGALoader sometimes reports errors

hedon77, 2023-10-15:
hi,

Compiled from sources... sometimes reports errors 

SDK 1.5.1 default configuration BOARD_TYPE BOARD PICO  USB_CDC_UART_BRIDGE  0

root@cnc-centrum:~/pico-dirtyJtag/build# openFPGALoader --cable dirtyJtag --detect
Jtag frequency : requested 6000000Hz -> real 6000000Hz
JTAG init failed with: Unknown device with IDCODE: 0xa0811043 (manufacturer: 0x021 (lattice), part: 0x04 vers: 0xa
root@cnc-centrum:~/pico-dirtyJtag/build# openFPGALoader --cable dirtyJtag --detect
Jtag frequency : requested 6000000Hz -> real 6000000Hz
JTAG init failed with: Unknown device with IDCODE: 0xf0001043 (manufacturer: 0x021 (lattice), part: 0x00 vers: 0xf
root@cnc-centrum:~/pico-dirtyJtag/build# openFPGALoader --cable dirtyJtag --detect
Jtag frequency : requested 6000000Hz -> real 6000000Hz
JTAG init failed with: Unknown device with IDCODE: 0xf8000043 (manufacturer: 0x021 (lattice), part: 0x40 vers: 0xf
root@cnc-centrum:~/pico-dirtyJtag/build# openFPGALoader --cable dirtyJtag --detect
Jtag frequency : requested 6000000Hz -> real 6000000Hz
index 0:
        idcode 0x1111043
        manufacturer lattice
        family ECP5
        model  LFE5UM-25
        irlength 8

phdussud, 2023-10-15:
This generally indicates that the connection between the pico and the FPGA isn't good enough to run at the JTAG frequency chosen. I have had good luck with braiding additional ground wires with the JTAG signals.

hedon77, 2023-10-15:
thx..  it started working properly, it's possible that there was a problem with the connecting cables... best regards

phdussud, 2023-11-04:
closing this one

closed 2023-11-04 by phdussud