phdussud/pico-dirtyJtag #19: Support for OpenOCD

abo-github, 2024-02-02:
The provided firmware works great and I establish communication between my pico and a board with an Samsung S3C2416 via UrJTAG. But unfortunatly I am not familiar with UrJTAG, but rather with OpenOCD.

But I can not find the right configuration to work with this pico.

Any suggestions? I guess it will not be compatible.

phdussud, 2024-02-02:
You can try this:
[https://sourceforge.net/u/phdussud/openocd/]
It hasn't been tested much so I can't guarantee anything

abo-github, 2024-02-03:
Oh yes! This worked like a charm. Many thanks 👍

closed 2024-02-03 by abo-github

phdussud, 2024-04-03:
I have another branch of openocd that I try to keep up to date with the official repo of openocd. It contains a new implementation of the dirtyjtag code. The major difference with the implementation that is on sourceforge concerns batching of requests so there is less USB traffic. I am not sure it ends up feeling faster when you debug but if you guys want to try it out, it is https://github.com/phdussud/openocd