phdussud/pico-dirtyJtag #21: Openocd Cfg File

invisible789, 2024-03-06:
Very nice project indeed, i'm looking for openocd Cfg file pico for debugging, can someone share ? i'm using already build uf2

phdussud, 2024-03-06:
You need more than a cfg because openocd bakes probe support code in the openocd image. 
You can try this:
[https://sourceforge.net/u/phdussud/openocd/]
It hasn't been tested much so I can't guarantee anything

phdussud, 2024-04-03:
I have another branch of openocd that I try to keep up to date with the official repo of openocd. It contains a new implementation of the dirtyjtag code. The major difference with the implementation that is on sourceforge concerns batching of requests so there is less USB traffic. I am not sure it ends up feeling faster when you debug but if you guys want to try it out,  it is https://github.com/phdussud/openocd

closed 2024-04-27 by phdussud

jonas-s-s-s, 2024-10-18:
> I have another branch of openocd that I try to keep up to date with the official repo of openocd. It contains a new implementation of the dirtyjtag code. The major difference with the implementation [...]

Doesn't work at all. 
```
$ lsusb
Bus 001 Device 016: ID 1209:c0ca Generic Jean THOMAS DirtyJTAG
```
```
$ sudo openocd -f /usr/local/share/openocd/scripts/interface/dirtyjtag.cfg -f /usr/local/share/openocd/scripts/target/bcm2835.cfg
Open On-Chip Debugger 0.12.0+dev-g81f929a97 (2024-10-18-09:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
bcm2835.cpu
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : dirtyjtag version 2
Info : clock speed 4000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: bcm2835.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: missing data from bitq interface
Error: 'arm11 target' JTAG error SCREG OUT 0x00
Error: unexpected ARM11 ID code
Error: [bcm2835.cpu0] Examination failed
Warn : target bcm2835.cpu0 examination failed
Info : starting gdb server for bcm2835.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
```
With the "official" port of DirtyJTAG openocd (https://review.openocd.org/c/openocd/+/7344) it at least detects the CPU:
```
Info : JTAG tap: rspi.arm tap/device found: 0x07b7617f (mfg: 0x0bf (Broadcom), part: 0x7b76, ver: 0x0)
```
With this it doesn't do anything.