phdussud/pico-dirtyJtag #22: cJTAG and OpenOCD

luyi1888, 2024-04-07:
Hi, I am trying to implement a cJTAG debugger on Pico. I'm already finished PIO part.
The protocol is not so complicated. It is a wakeup sequence and JTAG(TMS+TDO+TDI) transform to cJTAG(TMSC=TMS+TDO+TDI). So there is no difference to the upper.

Do you willing to merge it if I submit a PR?

If yes, Could you tell me the status of OpenOCD support? Where can i get the latest code and example config?
As far as I know, The cJTAG mainly used in RISC-V segment and some TI MCU. Most of RISC-V vendor have their own OpenOCD fork. And I want to use it on RISC-V MCU. I ask this question because I need to patch their code.

phdussud, 2024-04-07:
Well, I don't think we want to extend Pico DirtyJtag beyond its 100% compatible brother the original [DirtyJtag](https://github.com/dirtyjtag/DirtyJTAG). Your code would have to be working on the original DirtyJtag devices as well.  @jeanthom would need to be on board. Of course, you are welcome to fork this port if you change the USB descriptor product ID and relevant strings, so users won't get confused. 
About cJTAG: It seems to be somewhat of a minefield... See https://wiki.segger.com/J-Link_cJTAG_specifics 
You are also welcome to take my changes to OpenOCD and modify them to support your cJtag device. I have 2 implementations. 
A simple one on https://sourceforge.net/u/phdussud/openocd/ and more complicated one on a recent port on https://github.com/phdussud/openocd It is more complicated because it buffers commands when possible before sending them to the device. Based on stm32 device debugging, both implementations work. I am not sure if the more complicated one is any faster.

luyi1888, 2024-04-08:
Ok, Thank you for your advice.
I will try to fork this port first.

closed 2024-04-08 by luyi1888

jeanthom, 2024-04-08:
Hi,

I think cJTAG would be a great addition to DirtyJTAG. I would rather exclude protocols like SWD, which need very protocol-specific code running on the probe, and for which many excellent probe projects are already available and better supported in OpenOCD & other debugging software (like BMP or Raspberry's).