phdussud/pico-dirtyJtag #24: modify OpenOCD driver

luyi1888, 2024-05-08:
Hi, I am trying to make cJTAG happen.
The current dirtyJtag OpenOCD driver based on bitq interface. 
I need issue a command to dirtyJtag to send cJTAG escape sequence  in bitq_execute_queue(). It seems like it can't be done without heavily modify bitq.c.
 I can only think of rewriting dirtyJtag.c completely using  ftdi.c as template.
Is there a simple way to archive this?

That's what I want to archive
https://github.com/riscv-collab/riscv-openocd/blob/ae7ffa424e1b686559ae833ce2d384ded47ba4c4/src/jtag/drivers/ftdi.c#L672

phdussud, 2024-05-08:
The simplest would be to implement the reset field of bitq_interface and modify the JTAG_RESET like it is modified in ftdi.c

https://github.com/riscv-collab/riscv-openocd/blob/ae7ffa424e1b686559ae833ce2d384ded47ba4c4/src/jtag/drivers/ftdi.c#L672

luyi1888, 2024-05-21:
Do you know the reason why wrote this?
https://github.com/phdussud/openocd/blob/86931cb649b00d908652514d1f0ce9ae0996d03d/src/jtag/drivers/dirtyjtag.c#L439

I have to send IR code 0x10/0b00001, So it will encounter this.

I have rewritten code, It seems to be fine.
https://github.com/luyi1888/openocd/blob/22ffe21b757a6d69bbf6859dfe75f3ccf13f3c5d/src/jtag/drivers/dirtyjtag.c#L446-L458

luyi1888, 2024-05-21:
By the way, the initial cJTAG support is complete.
Anyone who interested please feel free to try.

OpenOCD repo: https://github.com/luyi1888/openocd
pico-dirtyJtag repo: https://github.com/luyi1888/pico-dirtyJtag
Demo usage: https://github.com/luyi1888/pico-dirtyJtag/issues/1

phdussud, 2024-05-21:
About the code you had to modify... I think it was an oversight. Good catch.

phdussud, 2024-05-21:
Just to be sure... Your version of the pico-dirtyjtag firmware is not compatible with any of the utilities using dirtyJtag (OpenFPGALoader, etc...) so people are going to be confused and won't have a way to identify what's wrong. 
I would strongly prefer that you change the string returned by the command cmd_info to something like DCJTAG. This way the software checking for the name won't be confused. 
This would simplify your changes of OpenOCD since you wouldn't need to have the subcommand.

luyi1888, 2024-05-22:
Ok. I change it to DCJTAG.

phdussud, 2024-05-23:
Great. Now you can test the name of the adapter in the opening code sequence of openocd and set your flag dirtyjtag_cjtag_protocol based on the adapter name

closed 2024-06-16 by phdussud