phdussud/pico-dirtyJtag #25: Problem with UrJTAG

AvatarBg111, 2024-05-31:
Hello, I'm trying to debug a STM32F103 microcontroller using a Raspberry Pico with pico-DirtyJTAG firmware. I successfully flashed the Raspberry and tried connecting to it using openFPGALoader and I think it worked, because I got some IDs of the chip (IDCODE: 0x00000003, manufacturer: 0x001, part: 0x00 vers: 0x0). But openFPGALoader prompted me that the chip was an unknown device for it. So I decided to try out urjtag, but there is one step in the README.md file of your project that I cannot execute - "cable dirtyjtag". There is no "dirtyjtag" in the list of "cables". How do I add it, because I can't find any source that shows adding a "cable" configuration?

jeanthom, 2024-05-31:
Hi,It seems like you are running a very old version of UrJTAG, or your build has dirtyjtag disabled. I'd recommend you rebuilding UrJTAG from source.Please be aware that UrJTAG is a horrible option for debugging modern ARM-based platforms. You'd better try the OpenOCD fork with DirtyJTAG support.Jean

AvatarBg111, 2024-05-31:
Ok, thanks for the help

closed 2024-06-16 by phdussud