phdussud/pico-dirtyJtag #26: error building the project: #include "jtag.pio.h", no jtag.pio.h file.

fsancet, 2024-06-16:
Building the project in vs code (clear build). Error while JTAG/pio_jtag.c:4:10: fatal error: jtag.pio.h: No such file or directory

Is this a tool configuration issue?

phdussud, 2024-06-16:
Yes it is. This file is supposed to be built by a tool that is itself build using a C compiler on the build machine. Look at your log (or output on the terminal) you should nee a problem trying to build the 2 tools necessary for the successful build of the pico executable. Note that if you are happy with the default there is an executable ready to download in the Releases section if my github repo

fsancet, 2024-06-17:
@phdussud, thanks for the hint. I'm not very expert of the C compiler
setup, and am familiarising now with vs code. Btw, what are the 2 tools you
are talking about?

Francesco

phdussud, 2024-06-17:
I am talking about PioasmBuild and ELF2UF2Build. Look at the beginning of my build log(end of message) for reference. 
I strongly advise you to read this [doc ](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) carefully. It contains detailed instructions to obtain a good build setup for most OSes

beginning of build log:
build] [  1%] No update step for 'PioasmBuild'
[build] [  2%] No update step for 'ELF2UF2Build'
[build] [  3%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[build] [  4%] No patch step for 'ELF2UF2Build'
[build] [  5%] Performing configure step for 'ELF2UF2Build'
[build] [  6%] No patch step for 'PioasmBuild'
[build] [  7%] Linking ASM executable bs2_default.elf
[build] [  8%] Performing configure step for 'PioasmBuild'
[build] loading initial cache file P:/pico/pico-projects/dirtyJtag/build/pico-sdk/src/rp2_common/tinyusb/pioasm/tmp/PioasmBuild-cache-Debug.cmake
[build] CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
[build]   Compatibility with CMake < 3.5 will be removed from a future version of
[build]   CMake.
[build] 
[build]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[build]   CMake that the project does not need compatibility with older versions.
[build] 
[build] 
[build] [  8%] Built target bs2_default
[build] [  9%] Generating bs2_default.bin
[build] [ 10%] Generating bs2_default_padded_checksummed.S
[build] [ 10%] Built target bs2_default_padded_checksummed_asm
[build] -- The C compiler identification is GNU 13.2.0
[build] -- The CXX compiler identification is GNU 13.2.0
[build] -- The CXX compiler identification is GNU 13.2.0
[build] -- Detecting CXX compiler ABI info
[build] -- Detecting C compiler ABI info
[build] -- Detecting CXX compiler ABI info - done
[build] -- Check for working CXX compiler: P:/msys2-64/mingw32/bin/c++.exe - skipped
[build] -- Detecting CXX compile features
[build] -- Detecting CXX compile features - done
[build] -- Detecting C compiler ABI info - done
[build] -- Configuring done (1.7s)
[build] -- Check for working C compiler: P:/msys2-64/mingw32/bin/cc.exe - skipped
[build] -- Detecting C compile features
[build] -- Detecting C compile features - done
[build] -- Generating done (0.1s)
[build] -- Build files have been written to: P:/pico/pico-projects/dirtyJtag/build/pioasm
[build] -- Detecting CXX compiler ABI info
[build] [ 11%] Performing build step for 'PioasmBuild'
[build] [ 10%] Building CXX object CMakeFiles/pioasm.dir/main.cpp.obj
[build] [ 20%] Building CXX object CMakeFiles/pioasm.dir/pio_assembler.cpp.obj
[build] [ 30%] Building CXX object CMakeFiles/pioasm.dir/pio_disassembler.cpp.obj
[build] [ 40%] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.obj
[build] [ 60%] Building CXX object CMakeFiles/pioasm.dir/gen/lexer.cpp.obj
[build] [ 70%] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.obj
[build] [ 70%] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.obj
[build] [ 80%] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.obj
[build] [ 90%] Building CXX object CMakeFiles/pioasm.dir/ada_output.cpp.obj
[build] -- Detecting CXX compiler ABI info - done
[build] -- Check for working CXX compiler: P:/msys2-64/mingw32/bin/c++.exe - skipped
[build] -- Detecting CXX compile features
[build] -- Detecting CXX compile features - done
[build] -- Configuring done (2.6s)
[build] -- Generating done (0.1s)
[build] -- Build files have been written to: P:/pico/pico-projects/dirtyJtag/build/elf2uf2
[build] [ 12%] Performing build step for 'ELF2UF2Build'
[build] [ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.obj
[build] [100%] Linking CXX executable elf2uf2.exe
[build] [100%] Built target elf2uf2
[build] [ 13%] No install step for 'ELF2UF2Build'
[build] [100%] Linking CXX executable pioasm.exe
[build] [ 13%] Completed 'ELF2UF2Build'
[build] [ 15%] Built target ELF2UF2Build
[build] [100%] Built target pioasm
[build] [ 16%] No install step for 'PioasmBuild'
[build] [ 17%] Completed 'PioasmBuild'
[build] [ 19%] Built target PioasmBuild

fsancet, 2024-06-17:
> [build] Update the VERSION argument value or use a ... suffix to tell

Thanks a lot. I'll double check and give you a feedback

phdussud, 2024-07-17:
No news is good news, I guess.

closed 2024-07-17 by phdussud

fsancet, 2024-07-17:
Hi, yes, this is closed