phdussud/pico-dirtyJtag #27: SIG_SRST cannot be controlled with BOARD_PICO configuration.

D4VF0, 2024-07-16:
Hey,

First of all, great work on the firmware. However, I've noticed that despite defining `#define BOARD_TYPE BOARD_PICO`, I am unable to control the `pin_rst` as "emulate open drain with pull up and direction". I observed that in the file pio_jtag.c, the lines 244 and 265:
`#if !( BOARD_TYPE == BOARD_QMTECH_RP2040_DAUGHTERBOARD )`
have no effect because #include "dirtyJtagConfig.h" is missing and thus, the information about BOARD_TYPE is absent.

By adding #include "dirtyJtagConfig.h" to pio_jtag.c, I was able to control the reset pin as desired by the function.

Thank you and best regards,
Dave

phdussud, 2024-07-17:
Hi Dave. Thank you very much for catching this problem. I will update the code base shortly.

phdussud, 2024-07-17:
Updated the code base.

closed 2024-07-17 by phdussud