phdussud/pico-dirtyJtag #33: BSR MSB sampled always as zero

jesari-git, 2025-01-27:
The MSB of the Boundary Scan Register is read always as zero after a shift-DR command when IR = SAMPLE/PRELOAD of EXTEST. (maybe the MSB of IDCODE is also wrong, but in my case it is 0, so we can't tell)
But, the TDO waveform in an scope screen shows the expected changes when we toggle the corresponding pin (being the last bit shifted it stays with the same level quite a long time), so, I guest there is a bug in the shifting code of the RP2040. 

The boundary scan register of my prototype was 30-bit long (I don't know if this matters) I tested this using URjtag + pico-Dirtyjtag

Regards

phdussud, 2025-01-27:
hello! 
Sorry for this problem. Can you tell me if the other bits are properly read? i.e. the shift that occurs when not reading a multiple of 32 bits works correctly. 
Thanks!

jesari-git, 2025-01-27:
Yes, the remaining 29 bits seems to be read with their correct values and at their proper places. Only bit #29 is stuck at 0.

Excuse me if I can't be more specific. I tried to look at the source code but it is just too complex for me (with PIOs and DMA...) And it is still possible to have the problem in the URjtag side...

phdussud, 2025-01-27:
I was looking at the code as well. There is no DMA in the picture when the transfer is less than 32 bits. This is the remaining code. 
void __time_critical_func(pio_jtag_write_read_blocking)(const pio_jtag_inst_t *jtag, const uint8_t *bsrc, uint8_t *bdst,
                                                         size_t len) 
{
    size_t byte_length = (len+7 >> 3);
    size_t last_shift = ((byte_length << 3) - len);
    size_t tx_remain = byte_length, rx_remain = last_shift ? byte_length : byte_length+1;
    uint8_t* rx_last_byte_p = &bdst[byte_length-1];
    io_rw_8 *txfifo = (io_rw_8 *) &jtag->pio->txf[jtag->sm];
    io_rw_8 *rxfifo = (io_rw_8 *) &jtag->pio->rxf[jtag->sm];
    //kick off the process by sending the len to the tx pipeline
    *(io_rw_32*)txfifo = len-1;
    {
        while (tx_remain || rx_remain) 
        {
            if (tx_remain && !pio_sm_is_tx_fifo_full(jtag->pio, jtag->sm))
            {
                *txfifo = *bsrc++;
                --tx_remain;
            }
            if (rx_remain && !pio_sm_is_rx_fifo_empty(jtag->pio, jtag->sm))
            {
                *bdst++ = *rxfifo;
                --rx_remain;
            }
        }
    }
    //fix the last byte
    if (last_shift) 
    {
        *rx_last_byte_p = *rx_last_byte_p << last_shift;
    }
}

phdussud, 2025-01-27:
I know nothing about URjtag. would it be possible to ask for 31 bits and see if the 30th bit is correct then? I am wondering if the length passed to Dirtyjtag is correct.

phdussud, 2025-01-27:
OK, something to try to check out the firmware and hardware of DirtyJTAG. 
Connect TDO(17) to 3.3V or TDI(16)  If we read 30 bits of TDO we should read 0xFFFFFFFC (being little endian)
It would be good to observe TCK and TDO with a scope to confirm something is happening to the HW
Run the following python script
```
import usb1
context = usb1.USBContext()
handle = context.openByVendorIDAndProductID(0x1209,0xC0CA)
handle.claimInterface(0)
buff = bytearray(b'\x01\x00') # CMD: ID
handle.bulkWrite(1,buff) #in endpoint is 1
result = handle.bulkRead(0x82,64) #out endpoint is 0x82
if not result.startswith(b'DJTAG2'):
    print('wrong version of jtag')
    exit()
command = bytearray(b'\x03\x1e\xff\xff\xff\xff') # CMD XFR, len=30, TDI all high
handle.bulkWrite(1,command) #in endpoint is 1
result1 = handle.bulkRead(0x82,64) #out endpoint is 0x82
print(result1) # should print bytearray(b'\xff\xff\xff\xfc')
```
This test works on my hardware....

jesari-git, 2025-01-28:
Hi,
let me include some captures:

sudo jtag

UrJTAG 2021.03 #d9a2943f
Copyright (C) 2002, 2003 ETC s.r.o.
Copyright (C) 2007, 2008, 2009 Kolja Waschk and the respective authors

UrJTAG is free software, covered by the GNU General Public License, and you are
welcome to change it and/or distribute copies of it under certain conditions.
There is absolutely no warranty for UrJTAG.

warning: UrJTAG may damage your hardware!
Type "quit" to exit, "help" for help.

jtag> cable dirtyjtag
jtag> detect
USB read failed (timeout expired ?)
IR length: 3
Chain length: 1
Device Id: 10000000000001000111111110101011 (0x80047FAB)
  Manufacturer: Generic manufacturer code (0xFAB)
  Part(0):      LARVATT (0x047)
  Stepping:     rev8
  Filename:     /usr/local/share/urjtag/nolabs/laRVaTT/laRVaTT
jtag> scan
jtag> dr
011111111111111111111111111111 (0x1FFFFFFF)
jtag>
Here the TAP was synthesized in a FPGA. The 32-bit IDCODE is OK (I changed its MSB to 1 for this test), but the BSR should be 0x3Fffffff (all input bits were 1)

Then I connected TDO to +3.3V and results were the same:
jtag> scan
jtag> dr
011111111111111111111111111111 (0x1FFFFFFF)

If TDO is connected to ground the value is 0x00000000 as expected.
Now, with a 31-bit BSR:
jtag> scan
jtag> dr
0111111111111111111111111111111 (0x3FFFFFFF)

Again, it should be 0x7Fffffff

And with 32-bit BSR:
jtag> scan
jtag> dr
01111111111111111111111111111111 (0x7FFFFFFF)

it should be 0xFFffffff. In all cases the MSB bit was 0. But, the MSB of the IDCODE register is OK, its width is now the same as that of BSR, and I see no reason for using a different code for its shifting.

And about the python code. If fails, probably due to my PC configuration...

sudo ./tst.py
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/426.
./tst.py: 2: Syntax error: "(" unexpected

This is quite confusing, but I hope it can give you some clue. Maybe the bug is just in the printing of DR by UrJTAG... I should test it with a different JTAG adapter just to see if the problem persist, but I have none of these at hand.
Have a nice day

jesari-git, 2025-01-28:
Hi again
I solved my phyton problems (libusb1 was missing) and I ran your script:

sudo ./tst.py 
bytearray(b'\xff\xff\xff\xfc')

And all the bits were retrieved without problems. So, now I think the bug is in the UrJTAG code.
Please excuse me for this bug report and thanks a lot four your support.

phdussud, 2025-01-28:
I am sorry about this problem. @jeanthom may have some insights. I think he wrote the dirtyJTAG interface for UrJTAG

phdussud, 2025-01-29:
Hello, 
I took a look at UrJTAG and it looks like my implementation for the picoDirtyJTAG isn't close enough to the original implementation of DirtyJTAG. It get exposed by UrJAG. I believe the last bit of transfer is handled specially and in my current implementation, the last bit of TDO is not captured. Could you flash your pico with this code contained in the enclosed zip file and see if it fixes your problem? 
Thanks! 

[dirtyJtag.zip](https://github.com/user-attachments/files/18593156/dirtyJtag.zip)

jesari-git, 2025-01-30:
Hi, these are the results with the old and new Firmware:
OLD:
sudo ./tst.py 
bytearray(b'\xff\xff\xff\xfc')
bytearray(b'\x00')

jtag> scan
jtag> dr
011111111111111111111111111111 (0x1FFFFFFF)

NEW:
sudo ./tst.py 
bytearray(b'\xff\xff\xff\xfc')
bytearray(b'\x08')

jtag> scan
jtag> dr
111111111111111111111111111111 (0x3FFFFFFF)

I also tried it with a different FPGA design and the MSB now shows activity.
So, I think the problem with MSB is finally solved. Thanks!!!

phdussud, 2025-01-30:
Fix in release V1.05

closed 2025-01-30 by phdussud