phdussud/pico-dirtyJtag #34: Related project

harbaum, 2025-02-20:
Thanks or this great project.

I've started something similar using the PICO as a ft2232 compatible device. This has the downside of re-using the FTDI vid/pid but on the other hand allows to use all the tools that expect a ts2232 which include the vast majority of vendor provided closed source tools.

See https://github.com/harbaum/PICO-MPSSE/tree/main/pico_mpsse

I am currently using pure GPIO bitbanging but I'll look into your PIO code as the Efinix Trion T20 FPGA I am promarily intend to use this with is very picky about the JTAG/TCK timing and clock and e.g. doesn't work below 6Mhz. You code seems a great basis for a more exact timing.

It seems you are blocking during JTAG transfer which would mean that USB stalls during JTAG transfer and vice versa. Have you considered using a double buffer and running PIO/JTAG in parallel with USB?

phdussud, 2025-02-20:
@harbaum Hello!
Thank you for your interest in the code. If you look closely at dirtyjtag.c, you will see that the USB servicing and the JTAG servicing (PIO) run concurrently on a different core core1_entry for JTAG and jtag_main_task for USB. Sorry about the name that is very misleading! There is a 4 deep FIFO used to communicate between the 2 cores. See line 43 of the same file

phdussud, 2025-02-20:
Sorry I didn't notice your edit. I did tune the depth of the FIFO based on a lattice FPGA programming. I used a LA to look at the USB traces and the JTAG traces and there is virtually no gap between the USB packets and no gaps between the JTAG bursts. More depth didn't improve the situation.

harbaum, 2025-02-20:
I will look a little deeper into your code. I still have to fully understand how you do the TMS transmission. It seems it's handled by grouping all same-tms-bit transmissions into one run and set TDI to a fixed value during that.

And yes, using both cores sure is one solution. My initial thought was to use a double buffered dma with one buffer written by USB while the other one was emptied by JTAG.  But I have to admit that I have not used Pio before and I really need to play around with it to understand how it works. Maybe TMS can be handled by the Pio as well ...

phdussud, 2025-02-20:
I think TMS wiggling could be handled by the PIO as well but there isn't an instruction for this in the DirtyJTAG spec. Jean Thomas, who created the original DirtyJTAG was pretty thoughtful about choice of commands. I have never seen TMS wiggling as a bottleneck to FPGA programming so that's probably why he didn't create a command like XFER but for TMS. I have looked at SVF files for lattice and Xlinx and I haven't found a case of TMS wiggling besides setup. The main programming task is all handled by TDI wiggling. I don't know anything about other FPGA programming. If you could share an SVF file for Efinix it would be helpful to me.

harbaum, 2025-02-21:
There is no fundamental difference to other FPGAs and TMS is only used a few times to enter the read/write IR/DR modes.  A SVF is depicted in e.g. https://www.efinixinc.com/docs/an038-programming-with-mcu-and-jtag-v1.2.pdf but that's rather unspectacular and Efinix themselves claim that the Trion T20 cannot be loaded from a SVF file. 

I am actually not sure about this and its my expectation that the SVF player simply aren't exact enough with their TCK timing. I am pretty sure I could load a SVF just fine if I had one.

Generating TMS also by PIO is less to achieve a certain speed but to meet the timing requirements as some devices seem to be rather picky even with clock rates being too slow.

Anyway, I have looked into your PIO code a little bit and I think it's a great source for me to start with. Why do you use the DMA only if at least 4 Bytes are to be sent? Doesn't it work below that threshold? And if that's the case, couldn't you just send the 32 bit length field via the DMA as well and would then always exceed the 4 byte limit and could thus always use DMA.

I still need to think about the TMS integration. One solution would be to use one bit of the counter value to differentiate between TMS and TDI shifting. But I don't think the PIO can do this kind of bit handling. Maybe it's possible to move only 31 bits into x and 1 bit into y and then branch on this y ... 

Another solution would be to always send stream of 2 bits into the PIO always containing TMS _and_ TDI and just out'ing then both at once. This would require more processing of the data sent into the FIFO and waste some buffer memory since only one of the toggles and the other one stays static all of the time.

phdussud, 2025-02-21:
If you look at the code closely, you will see that the TMS wiggling is also done with the PIO. The TMS is set before the clock starts its PIO timed cycle, which is OK by FPGA specs. What counts is the precise timing CLK cycle low-high-low. This is my understanding anyway.
About the DMA. Setting the DMA is slower than just sending the 4 bytes (one IO) to the PIO FIFO. It wouldn't make sense.

harbaum, 2025-02-21:
Are you sure about TMS? Isn't it just driven by the CPU as a normal GPIO here?

https://github.com/phdussud/pico-dirtyJtag/blob/0f4f0059378c424fb798174ae2ee20059cd58439/pio_jtag.c#L201

And yes, the DMA needs to transfer a few bytes to be effective. That's why I was suggesting to send the length field also via DMA. Then there's always at least 5 bytes to be sent and that may actually be quite effective.

phdussud, 2025-02-21:
It is. But it does not matter what happens while the clock is idle. The important thing is to respect the setup and hold time of every signal with respect to the clock.
Here the TMS get set while the clock is idle (low state) and when the PIO starts it will go through the normal clock cycle (low state, rise, high state, fall) with the correct timing.

https://github.com/phdussud/pico-dirtyJtag/blob/0f4f0059378c424fb798174ae2ee20059cd58439/pio_jtag.c#L201

https://github.com/phdussud/pico-dirtyJtag/blob/0f4f0059378c424fb798174ae2ee20059cd58439/pio_jtag.c#L201

phdussud, 2025-02-21:
Right the length field is also sent. You can try to always use DMA. I started this way, and my measurements led to this more complicated code. I don't think there is a significant impact either way.

phdussud, 2025-02-21:
I think using 2 outputs at the same time is a fine idea. Your buffer will contain both TDI and TMS values interleaved, right? That seems  fine. If your instructions can be broken into 2 phases (clock TMS with constant TDI and clock TDI with TMS constant) and you are willing to restart the PIO for each of these phases, you can cover both phases with 2 entry points leading to 2 different codes. This may avoid having to create a new buffer content with interleaved TMS TDI values. Not sure if this helps you...

harbaum, 2025-02-21:
Yes, that's exactly what I meant. I am interleaving the data. Works fine and is simple and allows me to let the pio engine just run.

In the other case I wouldn't even have to switch between to entry points. I have successfully used the msb of your 32 bit length value to switch between two paths. But I was unable to switch the output shift register between the two pins. The out always works on the same pin or pins.

phdussud, 2025-02-21:
In the other case, couldn't you have 2 fragments, like:
```
looptdi:
    out pins, 1     side 0      ; Stall here on empty (sideset proceeds even if instruction stalls, so we stall with TCK low
    nop             side 1      ; raise TCK
    in pins, 1      side 1      ; sample TDO
    jmp x-- loop    side 0
    push            side 0      ; Force the last ISR bits to be pushed to the tx fifo

looptms:
    out pins, 2     side 0      ; Stall here on empty (sideset proceeds even if instruction stalls, so we stall with TCK low
    nop             side 1      ; raise TCK
    in pins, 1      side 1      ; sample TDO
    jmp x-- loop    side 0
    push            side 0      ; Force the last ISR bits to be pushed to the tx fifo
```

harbaum, 2025-02-22:
In pio_jtag_write_blocking you are using a single byte 'x' as the DMA rx target. Isn't this overflowing if you write more than 8 bits?

phdussud, 2025-02-22:
That's correct.

phdussud, 2025-02-23:
Hey, I am curious about the USB framework you are using. It isn't part of the standard pico libraries, is it? As you noticed, I am using the tinyUSB which is part of the pico framework, but it is all synchronous as far as I can tell.

harbaum, 2025-02-24:
This isn't really a framework. It's based on the [low level example](https://github.com/raspberrypi/pico-examples/tree/master/usb/device/dev_lowlevel) of the pico sdk. It's rather simple and fully interrupt driven which seemed like a good starting point to me.

phdussud, 2025-02-24:
Good find! I don't look enough into these examples.

harbaum, 2025-02-26:
Thanks a lot for your explanations. I've released a first version of my project using the PIO as well. For now I am just using this interleaved TMS/TDI transmission and it works fine and looks simple and I think i'll keep it that way.

Thanks again!

phdussud, 2025-02-26:
Great. I will check it out. I am closing this thread. I will create an issue on your repo if I have some comments.