phdussud/pico-dirtyJtag #35: RP2350 Support

cyo-the-vile, 2025-04-03:
Trying to port this to RP2350.

Is it just pin definitions having to remain in PIO0 bank for this to be ported, or is there more to it than that to port this to RP2350?  So far I've 'ported' it with new pin assignments for the JTAG pins and openFPGALoader seems to detect it connected.  Trying to load firmware on a sipeed board now.

cyo-the-vile, 2025-04-03:
So far that is what I've done. 

#define PIN_TDI 10 
#define PIN_TDO 11
#define PIN_TCK 12
#define PIN_TMS 13
#define PIN_RST 26
#define PIN_TRST 27

With the following board.

[image]

So it isn't just changing the pins to pins in PIO0 bank, it has to do with offset?

phdussud, 2025-04-03:
yes. You should be ok since the PIO does not access the other pins. I would be conservative and keep PIN_TMS next to PIN_TCK. There is a case for controlling TMS with the PIO but that would be for the future.

phdussud, 2025-04-03:
By the way, it looks like your board pinout is almost the same as a pico 2 board. GP29 excepted

cyo-the-vile, 2025-04-03:
When using with openfpgaloader, it detects the chip fine.

[image]

It seems that openfpgaloader can detect the ported firmware as dirtyJtag fine. but I keep getting a TDO stuck error.  Uncertain if this is a porting issue or openfpgaloader issue.  

[image]

phdussud, 2025-04-03:
Also it seems that there are more changes needed. Please take a look at this thread. Also you should use the latest SDK (2.1.1)
[https://forums.raspberrypi.com/viewtopic.php?t=375753&form=MG0AV3&form=MG0AV3](https://forums.raspberrypi.com/viewtopic.php?t=375753&form=MG0AV3&form=MG0AV3)

phdussud, 2025-04-03:
Another red flag is the fact that TDO is pulled down. This tiggers a bug in the RP2350 which may lock up TDO if it is pulled down too weakly by the device you are trying to attach to the JTAG interface. See  errata RP2350-E9

phdussud, 2025-04-03:
Also there are 2 versions of the PIO API. You should add this on line 3 of jtag.pio
.pio_version 0 // only requires PIO version 0

cyo-the-vile, 2025-04-03:
I have added those code changes you have suggested and don't have different behavior. I also disabled pulldown for TDO.  I've also tried adding external pullup for TDO and have removed it. no changes.

phdussud, 2025-04-03:
I am trying to port to a pico2 board. So far, I have got the same result as you do. I see the TDI wiggling. So the problem is with the TDO -> PIO

phdussud, 2025-04-03:
I think I got it. add the last 2 lines in pio_jtag.c init_pins like so
```
static void init_pins(uint pin_tck, uint pin_tdi, uint pin_tdo, uint pin_tms, uint pin_rst, uint pin_trst)
{
    #if !( BOARD_TYPE == BOARD_QMTECH_RP2040_DAUGHTERBOARD )
    // emulate open drain with pull up and direction
    gpio_pull_up(pin_rst);
    gpio_clr_mask((1u << pin_tms) | (1u << pin_rst) | (1u << pin_trst));
    gpio_init_mask((1u << pin_tms) | (1u << pin_rst) | (1u << pin_trst));
    gpio_set_dir_masked( (1u << pin_tms) | (1u << pin_trst), 0xffffffffu);
    gpio_set_dir(pin_rst, false);
    #else
    gpio_clr_mask((1u << pin_tms));
    gpio_init_mask((1u << pin_tms));
    gpio_set_dir_masked( (1u << pin_tms), 0xffffffffu);
    #endif
    gpio_init(pin_tdo);
    gpio_set_dir(pin_tdo, false);
}
```

cyo-the-vile, 2025-04-04:
You nailed it! 

For reference, I had to pull TDO high with a 4.7k because all the vendors reference designs do that for their part.

[image]

closed 2025-04-04 by cyo-the-vile

phdussud, 2025-04-04:
The relevant part of the issue is explained in the RP2350 datasheet, section 9.3