phdussud/pico-dirtyJtag #36: How to set TDI without TCK?

nghfp9wa7bzq, 2025-05-20:
Hello!

https://github.com/phdussud/pico-dirtyJtag/blob/master/pio_jtag.h#L48

This essentially says that I have to toggle TCK if I want to set / change TDI.
Before this line, jtag_set_tms() function simply sets TMS, without TCK if I understand correctly.

This is very ambiguous, making it a bad software design choice IMHO.
I would prefer to be able to change only TDI without TCK - as it it done for other signals.

1. Why was it done this way?
2. Why have the 'same' functionality in CMD_XFER and CMD_SETSIG? See [cmd.c](https://github.com/phdussud/pico-dirtyJtag/blob/master/cmd.c).
3. How to set TDI without changing TCK?
4. Is there a low level API - that I didn't see - in dirtyjtag that allows essentially bit-banging ie. setting each signal individually?
5. If no, how can I implement that, preferably in a way that would be accepted as a PR?

No offense, please help.
Thanks.

phdussud, 2025-05-20:
1. Because the JTAG protocol is synchronous, flipping the signals are meaningless until the rising front of TCK. 
2. This is to ensure compatibility with the older version of the DirtyJTAG protocol.
3. It is not possible or interesting because of 1. I think if you want a device that can arbitrarily bit bang, look at this project; [https://github.com/harbaum/PICO-MPSSE/tree/main/pico_mpsse](https://github.com/harbaum/PICO-MPSSE/tree/main/pico_mpsse)
4. no, see 1.
5. please see 3.

closed 2025-05-20 by phdussud

nghfp9wa7bzq, 2025-06-18:
With all due respect, let me get back to this issue, now that I understand things a bit better.

> Because the JTAG protocol is synchronous, flipping the signals are meaningless until the rising front of TCK.

I wouldn't dare to call it meaningless.
In fact, you would have to set up TDI and TMS BEFORE raising TCK.

[image]

This is from a Xilinx datasheet ds187.
NOTE that I am not trying to split hairs, but I want it to be very clear, that ignoring setup times leads to bad results.
This is exactly why I've created this issue.
In another code, I had to add multiple delays for things to work and I have feared that I would have to do the same for p-dj.
Depending on setup (cable lengths and quality between host, pico and DUT and so on) that may still be the case for some.
So far p-dj has worked for me, probably because PIO is set up in a way that has sufficient setup time (for my setup).
Therefore the capability to change setup / hold time may be a valuable addition to the project...

> This is to ensure compatibility with the older version of the DirtyJTAG protocol.

OK, looking and [DJ cmd.c](https://github.com/dirtyjtag/DirtyJTAG/blob/master/src/cmd.c) this makes sense.

> It is not possible or interesting because of 1.

Looking at [jtag_set_tdi() in jtag.c](https://github.com/dirtyjtag/DirtyJTAG/blob/aa04b99ecdafe3f498f59a940143ed017ce3db67/src/jtag.c#L233) - there things are done exactly how I would expect them to be. Why is this repo so much different, if this should be compatible with DJ?
(ie.
DJ - set TDI without TDO
p-dj - set TDI only with TDO)

phdussud, 2025-06-18:
I think there is a problem with the following code, not the PIO code.
```
void jtag_set_tdi(const pio_jtag_inst_t *jtag, bool value)
{
    toggle_bits_out_buffer[0] = value ? 1u << 7 : 0;
}

void jtag_set_clk(const pio_jtag_inst_t *jtag, bool value)
{
    if (value)
    {
        toggle_bits_in_buffer[0] = 0; 
        pio_jtag_write_read_blocking(jtag, toggle_bits_out_buffer, toggle_bits_in_buffer, 1);
    }
}
```
The application code is supposed to set tdi then setclock but jtag_set_clk resets the TDI signcal before toggling the clock. 
I think the line ` toggle_bits_in_buffer[0] = 0;`  should be deleted.