phdussud/pico-dirtyJtag #38: Lots of errors logged with UrJtag + pictoDirtyJtag

JensRestemeier, 2025-06-03:
I am trying to program ATF150x devices with UrJtag. I used a Tigard (https://github.com/tigard-tools/tigard) board in the past without problems, successfully programming working chips. I am experimenting with pico-dirtyJtag as a cheap option for people wanting to replicate my project without buying an expensive programming cable.

I have SVF files, both produced by the official ATMISP tool, and by the fuseconv tool (https://whitequark.github.io/prjbureau/tools/fuseconv.html). Differences seem to be cosmetic, and both produce working chips with the Tigard adapter. No warnings or errors are reported during programming except for a warning about "ABSENT":

```
jtag> svf mapper_atmisp.svf progress
warning: unimplemented mode 'ABSENT' for TRST
detail: Parsing   3660/3668 ( 99%)detail:
detail: Scanned device output matched expected TDO values.
```

With pico-dirtyJtag I get an error for the RUNTEST instruction:
```
jtag> cable dirtyJtag
jtag> frequency 100000
jtag> detect
USB read failed (timeout expired ?)
IR length: 10
Chain length: 1
Device Id: 00000001010100000010000000111111 (0x0150203F)
  Manufacturer: Atmel (0x03F)
  Part(0):      ATF1502AS (0x1502)
  Stepping:     A
  Filename:     C:/msys64/mingw64/share/urjtag/atmel/atf1502/atf1502as
jtag> svf mapper_atmisp.svf progress
warning: unimplemented mode 'ABSENT' for TRST
error:   Set the cable frequency with 'FREQUENCY <Hz>'.
error: Error occurred for SVF command, line 11, column 0-21:
 RUNTEST.
detail: Parsing   3660/3668 ( 99%)detail:
detail: Scanned device output matched expected TDO values.
```
This did not produce a working chip and terminated quite quickly.

As a test I removed the runtest instructions from the SVF and reran the program, which produces a lot of spam along the lines:
```
detail: 1 combined into one (1 bits)
detail: flush(5)
detail: cutoff at n=0 because clock.tms=1 is unsuitable for transfer
detail: 0 combined into one (0 bits)
detail: flush(4)
detail: 1 clock(s)
detail: 1 clock(s)
detail: 10 transfer
detail: cutoff at n=3 because clock.tms=1 is unsuitable for transfer
detail: 3 combined into one (12 bits)
detail: in: detail: 0detail: 0detail: 0detail: 1detail: 0detail: 0detail: 1detail: 0detail: 1detail:
 0detail: 0detail: 0detail:
detail: out: detail: 0detail: 0detail: 1detail: 0detail: 0detail: 0detail: 1detail: 0detail: 1detail
: 0detail: 0detail: 0detail:
detail: flush(1)
detail: cutoff at n=0 because clock.tms=1 is unsuitable for transfer
detail: 0 combined into one (0 bits)
detail: flush(8)
detail: cutoff at n=0 because clock.tms=1 is unsuitable for transfer
detail: 0 combined into one (0 bits)
detail: flush(7)
detail: 1 clock(s)
detail: cutoff at n=1 because clock.tms=1 is unsuitable for transfer
detail: 1 combined into one (1 bits)
detail: flush(6)
detail: cutoff at n=0 because clock.tms=1 is unsuitable for transfer
detail: 0 combined into one (0 bits)
detail: flush(5)
detail: cutoff at n=0 because clock.tms=1 is unsuitable for transfer
detail: 0 combined into one (0 bits)
detail: flush(4)
detail: 1 clock(s)
detail: 1 clock(s)
detail: 9 transfer
detail: cutoff at n=3 because clock.tms=1 is unsuitable for transfer
detail: 3 combined into one (11 bits)
detail: in: detail: 0detail: 0detail: 0detail: 0detail: 0detail: 0detail: 1detail: 0detail: 0detail:
 1detail: 0det
```
Though surprisingly this produced a working chip.

This is using the dirtyJtag.uf2 release from yesterday (2nd June), and whatever urjtag that gets installed with msys2:
```
$ jtag --version

UrJTAG 2021.03 #d9a2943f
```

I am a bit unsure how to proceed, are the SVF files malformed and need to be fixed before programming, is urjtag broken, or is there a problem in in dirtyJtag? Unfortunately I don't have other cables or tools available that I could try.

[JsonToCupl.zip](https://github.com/user-attachments/files/20577198/JsonToCupl.zip)

juddbal, 2025-06-04:
I looked at this out of curiosity.  If you are able to recompile UrJtag, you might modify dirtyjtag.c: dirtyjtag_set_frequency() so that it sets cable->frequency = frequency; somewhere in the function body.

JensRestemeier, 2025-06-04:
FYI, Merge request on sourceforge: https://sourceforge.net/p/urjtag/git/merge-requests/11/
No idea how active they are, or if there is a more recent fork somewhere else...